rogerclarkmelbourne/OpenGD77 #857: Possiblilty to flash back to the original Radioddity firmware

michae1eo, 2020-11-16:
When I checking the updated user guide today, I found that some of the description that I have misunderstand.

The guide says:

>Installation of the firmware is undertaken at the owners own risk, but the official firmware can usually be reloaded onto the radio if the user has problems with the firmware, **as long as the operator takes a full backup of their radio using the OpenGD77 CPS immediately after the firmware has been installed** (see [*backup section*](#backup-before-you-do-anything-else))

I was thinking the **official firmware** is referring the Radioddity firmware which come with the radio originally, but it seems that we are talking about the OpenGD77 firmware here. In that case, does it means once you flashed the OpenGD77 firmware, it's impossible to flash back to the original firmware?

I noticed lots of chinese user try to flash the OpenGD77 firmware to their radio before learning the fact that it doesn't have chinese language support, thus urge to flash back the Baofeng/TYT firmaware with native chinese support, but to find there is no chinese firmware whatsoever.

rogerclarkmelbourne, 2020-11-17:
The original official firmware can be reflashed.

The User Guide says that this can be done only if the user makes a full backup of the radio.

Because the firmware uses parts of the Flash memory used by the official firmware, then the Flash has to be restored before the official firmware is reload, otherwise the things in the Flash memory which the official firmware uses, will be missing.

Actually, the Flash backup from any GD-77 / MD-760 would be OK. 

The backups from my radio, stored here would probably be OK

https://github.com/rogerclarkmelbourne/Radioddity_GD-77/tree/master/backups/GD-77

michae1eo, 2020-11-19:
Oh, thanks. That helps a lot.

closed 2020-11-19 by michae1eo