rogerclarkmelbourne/OpenGD77 #858: Jumping DMR slots

g7jiq, 2020-11-17:
Has anyone else noticed the GD-77 jumping slots on DMR.

I have a dual slot mmdvm both at home and in the car,
I had noticed it jumping from a programmed slot 2 to slot 1 when going into TX.
It will stay on slot 1 until another channel is selected or changed, and will return to slot 2 when the programmed channel is returned to.

I thought it was me, (catching the keys) but this happens on the second radio and dual mmdvm unit I use in the car.
Both mmdvm's work without problems using a tytera MD-380. 

Unfortunately this happens randomly and won't do it on command.

rogerclarkmelbourne, 2020-11-17:
FYI.

My MD-380 and GD-77 both running the official firmware would not both communicate with by duplex hotspot
I had to either adjust the hotspot settings to work with the MD-380 or adjust it to work with the GD-77.
The same applied to my Baofeng DM-1801 which required a different Tx / Rx offset to my GD-77 and to the MD-380

I discussed the problem with the MD-380 with Andy who write the MMDVM_HS firmware, and he said that often users need to change the modulation level for it to work with the MD-380.

g7jiq, 2020-11-18:
Hi, 
My MD-380 works 100%, and my GD77 has a better ber number and works fine , apart from jumping slots, which can be seen on the gd77 radios display, 
Its only on my dual mmdvm I notice this, as when it happens I appear on both my slots and loose my contact for that over.
As a solution I will swap over to a single channel mmdvm as channel slot doesn't matter. 
(as its intermitent I think this is the easyer fix)

jfml2020, 2020-11-20:
I realized that when listening on the same frequency, slot2 does not synchronize or if I put monitor mode, only slot1 is heard and if I force slot2 it is heard but choppy or as if it lost synchrony

pd4ajm, 2020-12-23:
Hello,
i'm using a Radioddity GD-77 with the last stable release, OpenGD77_R2020.10.16, and also have troubles with jumping time slots. 

Unfortunately also here it happens randomly and won't do it on command.

73", Toine PD4AJM

closed 2020-12-23 by g7jiq