rogerclarkmelbourne/OpenGD77 #861: Scan channels screen design

da3020, 2020-11-18:
Suggestion: it would be great to represent the Channel scanning screen in the way to avoid unreadable blinking of scanning channels names and parameters.
It's better and enough to indicate somsing like "Scanning Zone xxx" and some progress indicator during scanning.

vk7js, 2020-11-18:
Yes, and further to this, if you hit sk1, you hear the buzz of the channels and the voice prompt trying to announce the channel. It should just say scanning.

Thanks

Joe

g4eml, 2020-11-26:
I don't agree with this suggestion. 
One of the major comments from users of other DMR radios is that they do not display the current channel when scanning. 
 
Displaying the changing channel names while scanning is a really useful feature and gives the user confidence that the scan is in progress on the expected channels.

da3020, 2020-11-26:
Scan speed for analog and DMR channels is really different. With analog scan you really see nothing.

rogerclarkmelbourne, 2020-12-22:
This idea was tested but the response from anyone who tested it, was that it was a retrograde step. So this will not be implemented

closed 2020-12-22 by rogerclarkmelbourne