rogerclarkmelbourne/OpenGD77 #862: DMR BS data filtering

da3020, 2020-11-18:
Suggestion.
I am not an expert in DMR protocol, but I see that some repeaters periodically broadcast some special data that opens the squelch. I can see that data with my SDR as "BS data".
Radio reacts to that signals but it is not beneficial because of squelch opens, led and screen lights, scanning stops.
Maybe it possible to filtering that data.

rogerclarkmelbourne, 2020-11-18:
This is a known problem

Those repeaters are sending some roaming information which is a special data packet which the firmware does not currently handle

Unfortunately no repeaters in Australia transmit this data, nor do hotspots, so I have no way to simulate this , hence I can't analyse what data is sent and then stop it opening the squelch

The additional problem, is that its likely that this is manufacturer proprietary data, and Motorola may do something different to Hytera etc and the other manufacturers.

Although the DMR data spec is open, this is just the low level data transport level, and does not even include the audio codec or the SMS or other data formats, which are all closed source
So it may not be even possible to find any technical documents about what data is sent by commercial repeaters.

This is definitely the case for SMS, and although Brandmeister handles different manufacturers SMS formats, as long as you set the radio type in you Brandmeister SelfCare page.  Brandmeister is a closed source system, so we can't look at their source code to work out how to handle SMS etc

da3020, 2020-11-18:
Got it. The most I can help in this case it post IQ WAV file with our  provincial Hytera BS sounds.

rogerclarkmelbourne, 2020-11-18:
Unfortunately that won't help

You'd need to have decoded data frames.

da3020, 2020-11-19:
Some new ideas.

Maybe it seemed to me, but it looks like green LED is lit independently of squelch opening sometimes. I see the green led but no any sounds from speaker even in analog mode not to mention DMR with BS signals.
So, let the led lits when it wants, but use actual acoustic stream through the speacker as a trigger for stop scanning or torn the backlight on.

rogerclarkmelbourne, 2020-11-19:
Yes.

Yes. Green LED does not mean squelch open. It means signal detected.

If you have CTCSS on a freq the LED will illuminate even if the squelch does not open, also on DMR the LED will illuminate if there is a signal even if its not decoded.

This is normal functionality

da3020, 2020-11-19:
So -  now "Options - Display options - mode - squelch/auto" works not properly. That modes turn backlight on whem green led on but not ONLY IN CASE when squelch itself opens

rogerclarkmelbourne, 2020-11-19:
There may be some conflicts between those features.

The firmware is very complex and sometime a change in one place does not get fully updated in all the places which it impacts.

g4eml, 2020-11-26:
There is nothing special or unique about the beacon transmissions. They are simply a short transmission with both timeslots in the idle condition.  This is no different from what you would normally receive in the gap between two overs. So it would be difficult to ignore it when scanning.

rogerclarkmelbourne, 2021-02-14:
It would be impossible not to stop scanning for beacon transmissions, because it takes a considerable amount of time for the DMR signal to be analysed by the DMR chip.

I'm not sure of the exact number of DMR frames used to send these beacon transmissions, but the minimum time to detect any form of DMR signal is 120 ms, and at that point its not known what the signal contains.

So the scan would need to pause for at least 120milliseconds and possibly a lot longer.

Also, I'm not sure if these beacon frames are part of the ETSI standard. 
A lot of these things are manufacturer specific, so a beacon frame from Motorola may be some other transmission from Hytera etc.

closed 2021-04-20 by rogerclarkmelbourne