rogerclarkmelbourne/OpenGD77 #864: Band Limits: Off cannot be saved on the GD-77

n517rv, 2020-11-23:
When using OpenGD77 on the Radioddity GD-77 with versions above Stable Release Oct 16 2020, the Band Limits: Off option cannot be saved.  It reverts back to Band Limits: On.

"Downgrading" to Oct 16 2020 resolved the issue.

I tested this in the following releases and found this Band Limits Off save to be an issue:
Nov 12 2020
Nov 08 2020
Nov 03 2020
Nov 01 2020

Thanks for the fantastic OpenGD77 Firmware!

n517rv, 2020-11-23:
Not sure if it's actually a bug now because if you press the "Menu" button right away it will save the settings.  This isn't consistent across the various firmware versions.

rogerclarkmelbourne, 2020-11-23:
Are you pressing the PTT while in the Options screen before pressing Green to save the setting

n517rv, 2020-11-23:
No, it has to do with the fact that if you don't press "Menu" to get back
to the main screen it doesn't save the settings.  When I press "Menu" when
I'm on the Band Limits setting it will save it.  Not sure if this is by
design, but it's not consistent across the various releases.

I hope that clarifies things.  Thanks again for your development efforts!
Great software!

Steve
KA0AVG/AG

rogerclarkmelbourne, 2020-11-23:
You have to press the Green button to save any of the options or the Channel data changes.
And to make Channel changes permanent, you need to press SK2 + Green

This is the norm on all radio firmware's that I have ever seen.

Pressing Red cancels any changes, Pressing Green accepts / saves changes.

I guess we could add this to the User Guide, but currently there is a presumption of some standard concepts.

n517rv, 2020-11-23:
Ok, thanks for clarifying Roger.  Much appreciated.

Feel free to close this one out.

closed 2020-11-23 by rogerclarkmelbourne