rogerclarkmelbourne/OpenGD77 #869: TS2 not Displaying

VA3NDP, 2020-12-01:
RD-5r stops on active channel (during scan) that is set to TS2 and still displays TS1. The audio still comes through but it just doesn't display it. TS filter is ON as is CC filter.

f1rmb, 2020-12-01:
Is it a DMR simplex channel ?

VA3NDP, 2020-12-01:
No. It's a Repeater. I don't have any simplex Channels in my scan list. Thx

rogerclarkmelbourne, 2020-12-08:
Enable 

Display Settings -> Info to show the TS and tell me whether the TS indication is in bold or inverse or a different font.

VA3NDP, 2020-12-09:
"Inverse". 
I can change it manually per the instructions it just doesn't display TS 2 when it stops on a programmed channel using TS2. It works fine and it is picking up the correct TS but it just doesn't switch on the display. Thanks much !

f1rmb, 2020-12-09:
Hi,

  If the TS is in inverse video, it means you turned the TS filtering off, that's why it stops on any TS1 or TS2 channel.

VA3NDP, 2020-12-09:
Agreed. However, it doesn't change at anytime. Even when the TS filter is ON! That's why I was concerned. 
Cheers

f1rmb, 2020-12-09:
It will never change, as Roger told you, the displayed TS is the TX one, not one detected on RX (when TS filtering is OFF)

VA3NDP, 2020-12-09:
Yup. Got it. I was under the impression it was providing the "current incoming" TS. It's no problem if that's the way it was designed then it's perfect.

f1rmb, 2020-12-09:
Okay ;-)

rogerclarkmelbourne, 2020-12-22:
Closed as this is the normal operation of the firmware

closed 2020-12-22 by rogerclarkmelbourne