rogerclarkmelbourne/OpenGD77 #870: Squelch open delay

VA3NDP, 2020-12-01:
Redoing this as I had put it in the wrong Repo originally. 
RD-5R scans and stops on active channel and displays full signal,LED lights Green TG is displayed along with proper ID but audio doesn't come through at all or it takes a very long time to open the squelch. Even with TS & CC filters set to "On" and all info programmed correctly it just displays and hisses but will not play the audio or it takes 15-20 seconds to open. Radio right beside it programmed exactly the same way picked up or opened the squelch right away. I find I'm missing a lot. 
Thank you 
 [image]

rogerclarkmelbourne, 2020-12-03:
What do you mean open the squelch

There is no squelch on DMR.

VA3NDP, 2020-12-03:
I understand there is no squelch on DMR I just wasn't certain how else to say it. I mean when the actual voice comes through after it stops on an active channel. It will often stop on a working channel, display a full signal and a green LED and no voice can be heard right away (10-15 sec) or it doesn't open at all until the next person keys up. It even displays the TG and ID but no voice comes through. I can even hear the radio trying to play it with is small hiss noise along with all the indicators but again, no voice. I know it's a cheap radio. I will order a GD77 this week for comparison. I use mine to monitor a lot of DMR repeaters in my area ie: public works. We have a ton of DMR in use in our area so I am able to test quite a few aspects. Cheers!

f1rmb, 2020-12-03:
It's probably because you jumped in the middle of a transmission, so part of the DMR's "configuration" is missing, hence you need to wait to the next transmission, from the beginning, to be able to get the voice.

VA3NDP, 2020-12-03:
Open squelch for me you probably call it decoding audio. So that's what I meant if that helps. Having a radio right beside it that is talking away on the same frequency, TS and CC and the RD-5R display signal received and just will not "decode" the audio. Even at s9. _

VA3NDP, 2020-12-03:
> It's probably because you jumped in the middle of a transmission, so part of the DMR's "configuration" is missing, hence you need to wait to the next transmission, from the beginning, to be able to get the voice. 

I did not realize this was a thing so now that I know that is likely exactly what's happening because the next person that TX's the audio opens. Thanks. That's great to know. It's to bad it works that way but hey, it's DMR :) thanks again.

rogerclarkmelbourne, 2020-12-03:
Was this a simplex or a duplex signal ?

I can change channel onto an existing duplex signal from a duplex hotspot or a repeater and the radio normally decodes the DMR signal in less than a second.

Simplex does very slighly longer but not 10 to 15 secs

However, the time for the DMR chip start decoding the signal is dependant on a load of factors including the signal strength and how the was encoded in the first place.

So in theory you could have to wait a long time for the radio to start decoding the signal, but its extremely unusual for it to take as long as that.

The RF hardware is not at all good in the RD5R, and its very prone to co-channel interference or even being swamped by signals completely out of band, as it does not have separate RF stages for VHF and UHF. 

It just has a broadband front end, from 100MHz to about 600Mhz (and the same on Tx).

So I'd also not be surprised if some interference signal causing the DMR chip to have difficulties synchronising with the Rx DMR signal.

rogerclarkmelbourne, 2020-12-03:
As this problem seems isolated to your radio, and your operating environment I'm going to close this issue

Bear in mind, if this was a general problem with the firmware, someone would have reported it before, as the release on 7th May had 230 download for the RD5R and the 16th Oct version had 351 downloads, and no one else has this problem.

closed 2020-12-03 by rogerclarkmelbourne