rogerclarkmelbourne/OpenGD77 #871: Actual TS in TS filfer off mode

SQ9DKV, 2020-12-03:
suggestion: It would be good to have TS filter off displaying the timeslot the radio is currently receiving

rogerclarkmelbourne, 2020-12-03:
This would be inconsistent.

The display shows the Tx values not the Rx values

closed 2020-12-03 by rogerclarkmelbourne

SQ9DKV, 2020-12-03:
Yes I know. Therefore, I suggest that when the signal is received, ADDITIONAL information appears on which slot the reception takes place. Information about the TX slot is as it should be and I don't want to change it

SQ9DKV, 2020-12-03:
You can also make the TX TS change to the one that is being received. Then there will be no inconsistency

f1rmb, 2020-12-03:
Errr, no.
User case: you listening to a repeater, and both timeslots in use. How the hell could you start transmitting on the TS where the QSO you're interested of just freed-up?.

SQ9DKV, 2020-12-03:
It is about the situation when the TS filter is turned off. Then I listen to both TSs and while receiving the transmission the radio stops at the TS where the transmission is. However, it is not known which TS it is. One can only guess to answer correctly.

rogerclarkmelbourne, 2020-12-03:
The manual needs to be update, but if the TS on the screen does not match the currently received TS (or the TG does not match the currently received TG) the TG is displayed in inverse video

https://github.com/rogerclarkmelbourne/OpenGD77/blob/master/docs/OpenGD77_User_Guide.md#talkgroup-displayed-in-inverse-video

So if the display is in inverse video and you want to transmit on the same TS, press SK2

SQ9DKV, 2020-12-04:
In TS Filter off, always the TS indicator is displayed in negative. Regardless of which TS I receive the signal on. Therefore, I still do not see which TS the transmission is on.

rogerclarkmelbourne, 2020-12-04:
Yes. This is the correct funtionality.

RTFM

_In DMR mode, the current TimeSlot is shown to the right of the DMR text, e.g. TS2 for TimeSlot 2, and the Color Code, e.g. C1 for Color Code 1, is shown to the left of the battery voltage / percentage. When TimeSlot filtering) is turned off, the TS1 or TS2 indicator is in inverse color._

**I will close any more issues you raise  just because you can't be bothered to read the User Guide**

SQ9DKV, 2020-12-04:
I know it. I read the manual. You don't understand me. There is no information on what TS is transmitting on when TS filter is off. But if you don't want to enter this information, it's ok.