rogerclarkmelbourne/OpenGD77 #873: Analog ssquelch problem.

SQ9DKV, 2020-12-04:
When MD-1801 scans for All / Zone scan and battery is low, then (possibly under LOW BATTERY warning beep), the analog squelch stops working regardless of its settings. You have to turn off and on in and on the radio, but the problem comes back until it's charged.

rogerclarkmelbourne, 2020-12-04:
This is a hardware problem with this radio, because the DM1801 and RD5R don't use an audio chip with a mute control.
The mute control is via an external transistor and it doesn't work when the battery is flat.

Don't use the radio when the LOW BATTERY warning is displayed.

closed 2020-12-04 by rogerclarkmelbourne

SQ9DKV, 2020-12-04:
Wouldn't the option to disable beep on low battery be helpful? You may consider disabling this beep

trengking, 2020-12-04:
It is nice to make everything works as the way we want, unfortunately these are development builds and most of us just enjoying those for free. we should just be thankful and give valuable feedbacks that work for general use.

thankyou roger and team, please carry on the good work :)