rogerclarkmelbourne/OpenGD77 #877: Development release on the 12th November

da3020, 2020-12-07:
Here is some bug for 1801 in this release: quiet buz during Rx on two acoustic harmonics 875 Hz and 1750 Hz.

rogerclarkmelbourne, 2020-12-08:
I know of one other radio which has this problem, but because my DM-1801 does not have that fault, its impossible for me to debug what is causing it.

BTW. Are you certain of those exact frequencies (875Hz and 1750Hz), or is this an estimate of the frequency.
As 1750Hz is exactly the "Tone burst" frequency for old FM repeaters.

da3020, 2020-12-08:
> Are you certain

I see two peaks on acoustic spectrum analyzer quite precise: 875Hz and 1750Hz almost the same amplitude.
Also, these buzz disappears when holding monitor button.

rogerclarkmelbourne, 2020-12-08:
Is this on FM ?

Do you have any CTCSS on that channel

Monitor mode turns of CTCSS/DCS

rogerclarkmelbourne, 2021-03-02:
No reply from OP in nearly 3 months. Hence I'm closing this issue

Its possible it has now been fixed by various other changes made over the last 3 months e.g.

Try this version

https://github.com/rogerclarkmelbourne/OpenGD77/files/6072087/OpenGD77_20210303855.zip

closed 2021-03-02 by rogerclarkmelbourne