rogerclarkmelbourne/OpenGD77 #881: follow-on not honoured with rxctcss/txctcss  announcement on channel details

vk7js, 2020-12-21:
Hi Roger/Daniel,

This was something I also fixed in the older firmware. Now that you've fixed the announcement of rxctcss and txctcss etc in the channel details screen in your latest pre-release, you can't arrow rapidly to change the number because rxctcss and txctcss is spoken prior to the number as you rapidly arrow. I had fixed this so only the number was spoken when rapidly arrowing. Please let us be a beta tester to catch these things. In summary, follow-on needs to be honoured to allow rapid and efficient changing of codes in the channel details screen especially because there are 104 dcs and 50+ css codes.
Thanks
Joe
vk7js

f1rmb, 2020-12-27:
This has been changed in the latest beta.
Also, you can use SK2 + Left or Right, which jumps to the last/first CSS entry, then next/previous CSS type.

Cheers.

closed 2020-12-27 by rogerclarkmelbourne

vk7js, 2020-12-28:
Thanks for fixing this.

Joe

 

Closed #881 <https://github.com/rogerclarkmelbourne/OpenGD77/issues/881> .