rogerclarkmelbourne/OpenGD77 #885: Squelch tail on RX when using CTCSS

g7jiq, 2020-12-28:
I have just noticed when I am using ctcss on RX, I get a squelch tail.
(Turning the squelch up full makes no difference)
This wasn't present on ver D2020.12.22.01 or any before that.

This seems to be only present on the last two ver 27 and 28.
(Going back to ver 22 corrects the ctcss squelch tail.)

Would it be possible to tighten up the ctcss response as before.
Thanks

rogerclarkmelbourne, 2020-12-28:
The squelch tail when using CTCSS was extended after requests by several people who operate though repeaters which are very week and constantly fade in and out.

If the tail is reduced it would effect those operators, and my information is that the length of squelch tail that the firmware uses, is the norm for most ham radios.

g7jiq, 2020-12-28:
Thanyou for your reply, 
An easy solution for the problem of fading signal is to turn off the ctcss on rx only, this is what I have done in the past, and the result is what you have now.
But as it now stands the ctcss rx is useless, as I use my radio for allstar hubnet mainly when at home and I found it to be the best radio for monitoring when there is lotts of keyups, especially with the 50mw tx rf.
I now find the squelch tail annoying, 
Is it possible to make it switchable in the software or codeplug ?

g7jiq, 2021-01-19:
Problem has now been fixed 100%.
The squelch works as it should, with and without RX CTCSS.
Thankyou,
Well done...

closed 2021-01-19 by g7jiq