rogerclarkmelbourne/OpenGD77 #888: is there any guide to build?

jg1uaa, 2021-01-09:
Hello,

I tried to build OpenGD77 but I couldn't find building-how-to.
Here is my procedure, is there any better way?

**Requirements**
- arm-none-eabi-gcc 8.x or later
- newlib-nano
- mono

To match requirements, I installed Manjaro Linux (or Arch Linux) to virtual machine by VMware Player.

**Building**
```
sudo pacman -S base-devel git arm-none-eabi-binutils arm-none-eabi-newlib arm-none-eabi-gcc mono
(install all packages)
cd ~
git clone https://github.com/rogerclarkmelbourne/OpenGD77
cd OpenGD77/firmware/linkerdata
mono codec_dat_files_creator.exe
mkdir ../build
cd ../build
ln -s ../Makefile
make
```

That's all.

rogerclarkmelbourne, 2021-01-09:
See

https://github.com/rogerclarkmelbourne/OpenGD77/wiki/OpenGD77-SW-Development

Although someone added a Makefile via a PR, this is not the supported way to build. Its not been updated for a long time, and I should probably remove it, or at least move it to somewhere to indicate its not for general use.

Both Linux and Windows toolchain is the MCUExpresso IDE from NXP.

You do need to run the Windows exe to make the codec data files, but AFIK this will run under Wine etc

jg1uaa, 2021-01-09:
Oh, the how-to is at wiki page... I haven't read, sorry.

Indeed, I am an OpenBSD user and I want to build OpenGD77 on it.
But the version of arm-none-eabi-gcc of OpenBSD binary package is old (7.4.1) and no libc_nano.a, so it couldn't.

Using gcc-7.x produces "initializer element is not constant" error in some sources (for example, voicePrompts.c and sound.c).

I hope Makefile will be remained for adventurers, who cannot use or does not want to use MCUXpresso.

rogerclarkmelbourne, 2021-01-09:
There isnt a big problem using MCUExpresso, its free and although you have to make an account with NXP to download it, there are ways to protect your private information e.g. your email address and name e.g. yopmail etc

rogerclarkmelbourne, 2021-01-09:
BTW. IMO its impractical to do serious develop using the makefile and a separate editor

You could probably configure Eclipse to do what MCUExpresso does, as MCUExpresso is just a customised version of Eclipse, but it would be very hard work.

I use MCU Expresso and I have a JLink debugger soldered to the SWD pads on the PCB of the radio.

rogerclarkmelbourne, 2021-01-09:
I'll close this now, as I think the question has been answered

closed 2021-01-09 by rogerclarkmelbourne

jg1uaa, 2021-01-09:
Thanks a lot!