rogerclarkmelbourne/OpenGD77 #894: Does anyone want to support MMDVMHost's UserDB display?

jg1uaa, 2021-01-16:
Currently MMDVMHost can display UserDB information (user.csv) from RadioID.net with some displays, such as OLED and Surenoo TFT Display (Nextion has its own UserDB so not supported by MMDVMHost). Pi-Star's MMDVMHost does not support this feature yet but I hope it will be supported in near future.
Here is an old presentation at YouTube: https://www.youtube.com/watch?v=GAgk1JVEkDk

I also want to support UserDB display on OpenGD77 based radio.  Here is my idea:

1. Enhanced mode locked (default)
 Compatible with current implementation (except font size)

2. Enhanced mode unlocked
 MMDVMHost sends unlock command.
 After received, OpenGD77 displays UserDB information instead of calling updateContactLine()
 MMDVMHost simply sends UserDB information via MMDVM_SERIAL command.

To implement this feature, display layout need to be modified like this.
The different from current implementation, the font size of enhanced-locked mode will be smaller.
(I don't want to change font size between locked/unlocked mode because code will be fat.)

How OpenGD77 people think about this?

[image]

f1rmb, 2021-01-16:
Hi,

  Thanks, but no.
We already get the callsign when you're using the MMDVMHost's Cast driver, and that's way enough (this information is also embedded in the DMR data).
Doing such huge modifications in MMDVMHost and the Hotspot code looks a bit overkill for such a minimal benefit (who cares about such info on the screen of the Hotspot. Plus the fact it was quite complicated to make the hotspot code reliable and stable.

Cheers.
---
Daniel

jg1uaa, 2021-01-16:
Yes, current OpenGD77 uses DMR's talker alias to display callsign.
Indeed this is good enough, but I want to know someone wants richer experience or not so I put this issue.

Different from PC, OpenGD77 is an embedded system so I know that adding new feature carefully.
Many guys have same "no such feature!" opinion?

rogerclarkmelbourne, 2021-01-16:
I don't think this feature is worthwhile at the moment.

MCU program space and RAM is now limited, and to reserve space for major omissions like SMS, Power saving, Channel Dual watch, Dual Capacity Direct mode, Single Frequency repeater etc etc.

Even if you did write all the code for this feature for the hotspot, there is a chance that this would need to be removed at a later date to make space for more important features.
So would probably be a complete waste of your time.

jg1uaa, 2021-01-17:
OK, I asked here before coding so no problem.
Thanks for taking time for discussion. close this issue.

closed 2021-01-17 by jg1uaa