rogerclarkmelbourne/OpenGD77 #898: Station not respond to a personal call then DMR Filter is TG or RxG

R6DJO, 2021-02-09:
The station does not respond to a personal call if the DMR Filter is set to reception of talkgroup or reception list.

rogerclarkmelbourne, 2021-02-09:
Thanks

This does seem to be a bug.

I will be fixed

rogerclarkmelbourne, 2021-02-10:
[OpenGD7_Fix#898.zip](https://github.com/rogerclarkmelbourne/OpenGD77/files/5961775/OpenGD7_Fix.898.zip)

closed 2021-02-10 by rogerclarkmelbourne

R6DJO, 2021-02-11:
Thanks, everything works fine now/

rogerclarkmelbourne, 2021-02-11:
OK.

Note.
That version is my latest private build and also includes other bugs fixes for the "Motorboating" noise on DMR Tx, as well as the new "Eco Level" power saving feature, which will be automatically set to Level 1, which is a small amount of power saving when the radio is receiving and there is no signal.