rogerclarkmelbourne/OpenGD77 #899: Down arrow does not work after entering Options menu via QuickKey

rogerclarkmelbourne, 2021-02-10:
Report on the forum

https://opengd77.com/viewtopic.php?f=11&t=1530

f1rmb, 2021-02-11:
Hi Roger,

   I saw it on the forum, but completely forgot about it.
I will take a look at this today.

Cheers.
...
Daniel

rogerclarkmelbourne, 2021-02-11:
OK. I've not confirmed the bug.

I can probably fix it at the weekend.

Currently I'm still in the middle of investigating the official firmware power saving

f1rmb, 2021-02-11:
Okay. Well, I raised a PR for this one, so you can continue holding the probes ;-)

rogerclarkmelbourne, 2021-02-11:
Thankyou.

Yes. I'm still holding the probes..

Actually, I'm running the official firmware with the JLink attached, and using GDB to make changes to the GPIO manually, without the official firmware noticing ;-)

f1rmb, 2021-02-11:
LOL, just like a pickpocket.

rogerclarkmelbourne, 2021-02-12:
[OpenGD77_Fix_#899.zip](https://github.com/rogerclarkmelbourne/OpenGD77/files/5970142/OpenGD77_Fix_.899.zip)

closed 2021-02-13 by rogerclarkmelbourne