rogerclarkmelbourne/OpenGD77 #901: Pthon firmware loader not working

sm7tix, 2021-02-13:
stefan@stefan-MS-7681:~/OpenGD77/tools/Python/FirmwareLoader$ python3 gd-77_firmware_loader.py -m GD-77 -d U
 - Try to download the firmware for your GD-77 from the project page
 - Retrieve firmware location
Firmware download failed

Me or the file are wrong? :)
73 de SM7TIX / Stefan

f1rmb, 2021-02-13:
Does it fail silently ?
Have you installed the dependencies (take a look at the beginning of the file ?
Have you installed the udev file ?

sm7tix, 2021-02-13:
> Does it fail silently ?
> Have you installed the dependencies (take a look at the beginning of the file ?
> Have you installed the udev file ?

As you see in my text it stops with "Firmware download failed" 
It worked before, but today it didnt. So i clone the repo again but the same result. I think the modules are in place otherwise i have got a different error i think. The udev rules i dont think i have installed. I run as root so does it matter? Last time there was a new dev firmware i was able to download it.

sm7tix, 2021-02-13:
All dependencies are in place

f1rmb, 2021-02-13:
Hi,

 I just tried:

```
daniel@daniel-ThinkPad-X230:~ $ ./gd-77_firmware_loader.py -m GD-77 -d -U
 - Try to download the firmware for your GD-77 from the project page
 - Retrieve firmware location
 - Downloading the firmware version D2021.01.19.01, please wait
 - Now flashing your GD-77 with "/tmp/rmzeewph.sgl"
 - Firmware file confirmed as SGL
 - Sending Download command
 - Sending ACK
 - Sending encryption key
 - Sending F-PROG command
 - Sending radio modem number
 - Sending radio modem number 2
 - Sending version
 - Sending erase command
 - Send post erase command
 - Sending Program command
 - Sending last block                   
Firmware update complete. Please reboot the GD-77.
```

Your command line have a typo (as displayed, the dash is missing for the **U**.

sm7tix, 2021-02-13:
OK! My fault. Working now. You can close it with a remark of a stupid user :)

f1rmb, 2021-02-13:
Okay, cool.

sm7tix, 2021-02-13:
Voice prompt not chooseble after upate. Do  i have to update voice prompt too? Is there a pythonscript for that too?

f1rmb, 2021-02-13:
I think on the forum there is something.

sm7tix, 2021-02-13:
Ok.

rogerclarkmelbourne, 2021-02-13:
Does this issue need to stay open?

It appears to have been resolved

f1rmb, 2021-02-13:
Yes, it can be closed.

sm7tix, 2021-02-13:
Pls close

rogerclarkmelbourne, 2021-02-13:
I already closed it.

I think also you can close an issue if you opened it.

closed 2021-02-13 by rogerclarkmelbourne

sm7tix, 2021-02-13:
Ok!
Tnx.