rogerclarkmelbourne/OpenGD77 #905: Power Cycling RD-5R

VA3NDP, 2021-02-15:
RD-5R build Feb 13 2021
7098a74
Voice Prompts are ON
TS Filter- OFF
CC Filter - ON
DMR Filter - None
Filter - None
Eco Level - 0 (also tried it at all other levels)

Since the last 2 Firmware builds the unit has been Power Cycling (on/off) on its own. It cycles during monitor a single channel or scanning. It also happens at any power level. 
No other issues noted since the upgrade other then the constant power cycling.. 
Private calls seem to be coming through more often now as well. 

Hope i put this in the correct spot.
Thanks so much,
Nick
VA3 NDP

rogerclarkmelbourne, 2021-02-15:
Thanks.

What was the last version where it didn't reboot ?

VA3NDP, 2021-02-15:
The release from December 29th was ok.
D2020.12.29.02 beta (second development release December 29th).
thanks Roger

rogerclarkmelbourne, 2021-02-15:
OK.

This will need to go onto the ToDo list, because the Eco Levels seems to not work for about 5% of radios :-(

And there is a DMR Tx frequency calibration problem with the CPS which need to be fixed urgently.
And there is a DMR Tx power envelope issue on the RD5R which also needs to be investigated and fixed

VA3NDP, 2021-02-15:
No worries at all. I wanted to make sure I notified you about it. It's not that intrusive, I just restart the scan after the cycle and it's good until it cycles again. Sometimes it will go 10 min and other times it will go 60 min. It's all over the place. The only time it's a real problem is when I forget to turn the scan back on after it cycles. 😉 
Take your time with this one. It's not urgent like the other concerns you are dealing with. 

👍👍😀

VA3NDP, 2021-02-15:
New info- I've shutdown the voice prompts entirely (now beep only) and it HAS NOT power cycled once since I shut it off. It's been running on scan for hours now without issue. I haven't been able to get it to cycle at all. I will leave voice prompts off for the time being. Thanks much.

rogerclarkmelbourne, 2021-02-16:
Ah OK.

We have some issues with voice prompts occasionally causing a crash if they are used too vigorously.

e.g.  going up and down the menus quickly seems to occasionally cause a crash.

I think if you are leaving it scanning for ages, with the voice prompts on, the number of times its going to be announcing over a long period of time, is going to be a lot.
So even if it only crashes 1 / 1000 announcements, it will fail...

You mentioned that when it crashes, it ends up constantly rebooting.

Thats a strange one.

The RD-5R is not very good hardware, and I'd recommend you buy a DM1801 or even better a GD77.

The problem with the RD5R, apart from not very good RF sections, is that it does not have the power management control hardware.

And because of this, the radio can't do an orderly shutdown  and save the current configuration when the radio is turned off.

This forces the firmware to have to save any change to the radio, immediately to the EEPROM.

And I'm now wondering if perhaps its doing that when scanning, i.e around 30 times a second, which would not be good as it will "wear" the EEPROM, which although its spec'ed to allow millions of Writes, if you scan at 30 channels per second for days or months on end, I guess that would add up to a lot of Writes

rogerclarkmelbourne, 2021-02-16:
Yikes

At 30 channels per second, this is not good.

That's 30 x 3600 writes per hour. i.e over 100k writes per hour.

rogerclarkmelbourne, 2021-02-16:
We probably need to do something to suspend setting Writes when scanning, assuming its currently doing this at all, which I can't be sure of, as the firmware is changing the channel, not the operator.

VA3NDP, 2021-02-16:
Makes perfect sense. I will have to order myself a 1801. It's been scanning for hours now without a single reboot. Shutting off the voice prompts definitely did the trick. It's talking all the time and scanning 72 DMR channels without issue now. I'm fine with leaving the voice prompts off. I just love having it available when I show folks. 😀

rogerclarkmelbourne, 2021-02-16:
The voice prompts do push the limits of the hardware, I encode the prompts to AMBE using the radio its self to save space, but even using AMBE compression, it takes over 100k for the data.

I'm not sure why the playback occasionally crashes, its so intermittent that its difficult to debug.

rogerclarkmelbourne, 2021-02-16:
BTW. This problem may occur on the GD77 and DM1801 if left scanning with VP enabled, but I've not had any reports of that.

However, possibly people put up with it crashing from time to time, or perhaps they don't notice.

Definitely scanning with VP off is stable, as I know some people who leave their radio scanning 24/7 and it doesn't crash

VA3NDP, 2021-02-16:
To me it's a minor issue. You all have done so much work and an incredible job with this that I couldn't possibly complain lol. It's just so much better then the OEM firmware!! If I have to live with it, so be it. Cheers

VA3NDP, 2021-02-16:
I noticed it only because I have the restart Beeps (code) enabled and it stops my scan when it restarts. If I would have just left it on a single channel or repeater (not scanning) and I wasn't utilizing the startup beeps I wouldn't notice it at all because it's so quick. Having to restart the scan is the only real problem with it cycling. Sometimes I forget to restart the scan ans I miss a ton of calls ans info.

VA3NDP, 2021-02-16:
I can say without reservation that the last update definitely has given me almost double the battery life. I used to be able to scan about 10-12 hours straight with voice coming through on and off throughout this time frame. I'm now able to get two full scan days of 12 hours each. I used to charge the unit every night but now I can charge it every two nights and that's with eco set to 0. Whatever was done has helped tremendously with bitterly life. I noticed this even before I turned off the VP's.

rogerclarkmelbourne, 2021-02-16:
Yes.

During scanning it does not do any specific power saving, as thats done by turning the Rx and other chips in the radio On and Off.

But I made a load of other changes, which resulted in the radio taking a lot less power all the time.

Things like putting the Flash memory chip into its suspend mode, saves 7mA, and a whole bunch of other rather complicated power optimisations internally to the microcontroller chip.

rogerclarkmelbourne, 2021-02-16:
The Eco modes extends this further, but there seems to be some problems with some batches of the C6000 DMR chip, that don't support its sleep mode :-(

Seems OK on my RD5R and my GD77, and even with 300 downloads only 2 people said they had a problem .

f1rmb, 2021-03-06:
Hi,

  Which voice level are you using ?

xcvista, 2021-03-09:
About that EEPROM write, how about implementing a delayed write mechanism - regardless what caused the configuration change, the EEPROM would only be written to after a few seconds of configuration inactivity. This saves EEPROM on both automatic and manual scanning (smashing the up/down button to flick through channels rapidly.)

f1rmb, 2021-03-09:
@xcvista Do you really think the firmware is writing instantly all the data to the EEPROM ? 
No , it doesn't work like this, writing cycles are already managed and saved.

VA3NDP, 2021-03-09:
> Hi,
> 
> 
> 
>   Which voice level are you using ?
> 
> 

It was doing it at all levels for me. I used the first option the most thought. Since shutting off voice prompts I haven't had one reboot.