rogerclarkmelbourne/OpenGD77 #908: Squelch Tail is Back

g7jiq, 2021-02-21:


rogerclarkmelbourne, 2021-02-25:
There were a load of changes to various parts of the firmware, and I agree this bug probably has re-appeared.

I won't bore you with the technical details, but it can be fixed, as its just a hard coded setting which needs to be adjusted.

I'll log this as a bug.

Unfortunatly at the moment, a nasty bug has reappeared in the code and the RD5R and DM1801 versions are not working correctly on DMR. 

So I will not be able to release a fix for this until I can fix the DM1801 and RD5R bug, which unfortunately seems to be a tough nut to crack, as I've already been working  on the bug for well over a week, with no progress :-(

g7jiq, 2021-02-25:
This is not a problem, 
I understand there are more important problems to sort out.
I was just bringing this to your attention so it could possibly be looked at later.

rogerclarkmelbourne, 2021-02-25:
OK.

Thanks for letting me know.

I don't normally use the radio for FM, so often don't noticed a problem on FM unless its completely broken ;-)

rogerclarkmelbourne, 2021-03-03:
[OpenGD77_202103031642.zip](https://github.com/rogerclarkmelbourne/OpenGD77/files/6073815/OpenGD77_202103031642.zip)

I have reduced the tale to 50%, but you should not be relying on CTCSS to close the squelch.

You should set the squelch with CTCSS disabled and then re-enable CTCSS.

Its not possible to reduce the CTCSS hold time much further, as this has a negative effect on signals which are overdeviated.

closed 2021-03-28 by g7jiq