rogerclarkmelbourne/OpenGD77 #909: Backlight tip

Simon442, 2021-02-24:
Not sure if it's possible but I would like to see if backlight only came on when you press buttons on keypad, Quick menu button and both of the 2 side buttons and stay off when you speak or receive. Manual is something similar but you have to press Black key to turn it on/off

rogerclarkmelbourne, 2021-02-25:
This will be in the next release, but it may be several weeks or possibly longer as some nasty bugs have appeared in the DM1801 and RD5R versions

rogerclarkmelbourne, 2021-03-02:
https://github.com/rogerclarkmelbourne/OpenGD77/files/6072087/OpenGD77_20210303855.zip

Simon442, 2021-03-09:
> See
> 
> https://github.com/rogerclarkmelbourne/OpenGD77/files/6072087/OpenGD77_20210303855.zip
> 
> Includes a new backlight mode in Display Options , of "button"

Could it also be done so it doesn't turn on when you press PTT ?

rogerclarkmelbourne, 2021-03-09:
If you don't want the backlight on at all, just set it to off completely.

Its not practical to have a zillion different permutations of which buttons and screens do or don't have the backlight turned on.

Simon442, 2021-03-09:
> If you don't want the backlight on at all, just set it to off completely.
> 
> Its not practical to have a zillion different permutations of which buttons and screens do or don't have the backlight turned on.

That's true. Keep up the good work, looking forwards to more updates.

closed 2021-04-20 by rogerclarkmelbourne