rogerclarkmelbourne/OpenGD77 #910: Tone scan doesn't work when Filter is set to None

dmitrybabeshko, 2021-03-01:
Hello,

**Ver:** D2021.01.19.01

Steps to reproduce:
1. Set **Filter** to **None** in Quick Menu
2. Perform **Tone Scan**
3. Press **PTT** on the second radio with TX tones **disabled**.

_Actual:_
Tone scan stops immediately after the squelch is opened and displays the tone currently being scanned as found.

This is not an obvious behavior. Is it possible to temporarily set **Filter** to **CTCSS|DCS** or to display a warning message telling that **Filter** is disabled?

Thank you in advance!

rogerclarkmelbourne, 2021-03-02:
OK. 

I'll investigate

rogerclarkmelbourne, 2021-03-02:
Try this version

[OpenGD77_fix_tone_scan.zip](https://github.com/rogerclarkmelbourne/OpenGD77/files/6065991/OpenGD77_fix_tone_scan.zip)

rogerclarkmelbourne, 2021-03-02:
[OpenGD77_20210303855.zip](https://github.com/rogerclarkmelbourne/OpenGD77/files/6072087/OpenGD77_20210303855.zip)

dmitrybabeshko, 2021-03-02:
I have checked this one _OpenGD77_fix_tone_scan.zip_ on DM-1801. Looks like it works now.
Is there any difference with the last one?

rogerclarkmelbourne, 2021-03-02:
Last one has all other updates, including the tone scan fix.

FYI.
During Tone Scan, the filter FM filter is set to CTCSS (as this is necessary to detect a tone)
The filter setting is restored after the Tone Scan is complete.

closed 2021-03-02 by rogerclarkmelbourne

rogerclarkmelbourne, 2021-03-03:
https://github.com/rogerclarkmelbourne/OpenGD77/files/6073815/OpenGD77_202103031642.zip

dmitrybabeshko, 2021-03-05:
> https://github.com/rogerclarkmelbourne/OpenGD77/files/6073815/OpenGD77_202103031642.zip

With this firmware it started to work exactly as described in first message again.