rogerclarkmelbourne/OpenGD77 #915: A couple of possible slight bugs after last update

g7jiq, 2021-03-06:
These are only observations and feedback, 
They don't interfere with the main working of the radio.

There is no LCD backlight when radio is first turned on, when backlight mode is set to button, (ok on other settings).
The backlight is set to 1% or (above) Min brightness and brightness set to 10 (or any level) 

Second slight bug, 
The keytone disappears after a few mins when Eco Level is set above 1, This only happens on 1 of 2 radios (Both Radioddity)

rogerclarkmelbourne, 2021-03-06:
The keytone problem seems to be a hardware fault in some C6000 chips where the power saving feature of the chip seems to cause intermittent loss of beep audio.

Extra power saving is enabled on level 2 and above.  
This is documented in the User Guide, and is the reason the default power saving is level 1.

Level 2 and above, save 7mA more than Level 1, which is 25% of the current taken by the radio.
So I can't disregard this saving just because not all C6000 chips support it (at random)

rogerclarkmelbourne, 2021-03-06:
Re: 
_There is no LCD backlight when radio is first turned on, when backlight mode is set to button, (ok on other settings).
The backlight is set to 1% or (above) Min brightness and brightness set to 10 (or any level)_

Please can you list step by step in bullet points how to replicate this issue

f1rmb, 2021-03-06:
Hi,

  About the Button backlight mode, as it's name means, it's based on buttons events, only. So there is no point to trigger the backlight while booting.

Cheers.
...
Daniel

rogerclarkmelbourne, 2021-03-06:
Yes. Some people do not want any light from the radio. So it makes sense to not display while booting if button mode is enabled 

e.g. Nighttime use

rogerclarkmelbourne, 2021-03-06:
BTW. We can try sending those 96 bytes of 0xAA to OpenMusic when the radio exists deep sleep, but EA5SW said that didn't fix the beep problem for him, and only a total hardware reset of the C6000 made the beep work again after the C6000 was put into sleep mode :-(

f1rmb, 2021-03-06:
That C6000 bug is annoying. For sure. But the power saving is a huge gain.
Maybe adding something in the manual, for the people owning those buggy C6000, are in the need to set Eco level to 1.
Anyway, even at Eco 0, the power saving is already huge, compared to older code (clock etc...).

rogerclarkmelbourne, 2021-03-06:
I already updated the manual.

Perhaps I should put it in Bold.

But people never RTFM ;-)

f1rmb, 2021-03-06:
Yeah, and we cannot fix people, just blaming them for not reading the manual ;-)

rogerclarkmelbourne, 2021-03-06:
The only way to totally solve it would be a popup message.

But I will try posting a version here which sends the 0xAA when the firmware exists from deep sleep, just in case it fixes the problem for some people

rogerclarkmelbourne, 2021-03-07:
[OpenGD77_20210307_send_0xaa.zip](https://github.com/rogerclarkmelbourne/OpenGD77/files/6096427/OpenGD77_20210307_send_0xaa.zip)

@g7jiq

Attempt to fix the C6000 hardware bug. But I doubt if it works

EA5SW, 2021-03-07:
This version appears that works best in my GD77 .. testing now.....

EA5SW, 2021-03-07:
After more of 9 hours receiving and trasmitting with eco in level 4 this version works perfect in my GD77 at least in DMR Mode and beep  (not voice) in prompt.

PivoWro, 2021-03-07:
On my DM-860 testing for a few hrs with ECO level 2 works perfectly even beep is on DMR (buttons, dmr start, dmr stop)I will check on level 3 and 4 later on.

rogerclarkmelbourne, 2021-03-07:
OK.

The version in the zip is a bit of a sledgehammer to crack a nut, as it sends the re-sends the critical data to the C6000 many times a second.

I will need to do a better version which only sends the critical data when the radio finally exits from deep sleep, i.e when a signal is received.

But I will wait for some more people to test this version before I make any changes

PivoWro, 2021-03-08:
I'd like to confirm all 4 eco levels after few more hrs of testing ok. Beep doesn't disappear - MD-860 (0xaa)

rogerclarkmelbourne, 2021-03-08:
@PivoWro 

OK.

I need to do another version, which is a bit cleaner than this version.

I'll investigate how to do that later, and post another zip file here.

rogerclarkmelbourne, 2021-03-08:
OK

This version powers down the C6000 on all Eco modes > 0

It also only sends the Beep fix data to the C6000 when the Rx is receiving a signal

[OpenGD77_20210308_beep_fix_v2.zip](https://github.com/rogerclarkmelbourne/OpenGD77/files/6099288/OpenGD77_20210308_beep_fix_v2.zip)

Hopefully this version will work as well as the previous test version, and requires less data to be sent to the C6000

PivoWro, 2021-03-08:
Yup! I Can confirm beep fix v2 works good on eco level 4 - dmr stable and beep all the time is present. Tested on DM-860.

EA5SW, 2021-03-08:
Downloading.....

rogerclarkmelbourne, 2021-03-08:
@PivoWro 

All Eco levels now power down the C6000 chip, not just 2 , 3 and 4

Level 4 has long latency but the least current consumption.
In general levels 2 and 3 are probably the best to use, unless you need very long battery life.

PivoWro, 2021-03-08:
That's why I'm testing on the "hardest" level 4. Didn't notice so much latency - if level 4 works, other levels (3,2,1...) must look promising. Beep is present. Tested on my DM-860 device.

rogerclarkmelbourne, 2021-03-08:
@PivoWro 

OK. If level 4 works for you, the batter will have the longest life. 

Battery life will be longer than the official firmware, because the official firmware does not use the sleep function in the C6000.
I think they probably don't use the sleep feature because of the hardware bug in older radios.

BTW. 
How old is your radio which has this bug ?

PivoWro, 2021-03-08:
Well, radio I've bought in Jan'21, so to be honest don't know how old is... But on firmware D2021.03.06.01 eco level more than 1 DMR is not stable and beep is missing. With firmware (beep fix v2) looks alright both dmr and beep.

rogerclarkmelbourne, 2021-03-08:
I had a report that possibly the V2 bug fix didn't always work, so I've done a V3 which sends more reset data to the C6000

[OpenGD77_20210309_beep__fix_V3.zip](https://github.com/rogerclarkmelbourne/OpenGD77/files/6104350/OpenGD77_20210309_beep__fix_V3.zip)

rogerclarkmelbourne, 2021-03-08:
@EA5SW 

What are the results of your tests ?

PivoWro, 2021-03-09:
> 
> 
> I had a report that possibly the V2 bug fix didn't always work, so I've done a V3 which sends more reset data to the C6000
> 
> [OpenGD77_20210309_beep__fix_V3.zip](https://github.com/rogerclarkmelbourne/OpenGD77/files/6104350/OpenGD77_20210309_beep__fix_V3.zip)

Tested and is all ok.

rogerclarkmelbourne, 2021-03-09:
@EA5SW 

OK

EA5SW, 2021-03-10:
V3 as fine that V2, some times beep don't sound and also lost some DMR transmissions, v2 works more better for me.

rogerclarkmelbourne, 2021-03-10:
@EA5SW 

Is V2 perfect ? Or does it still sometimes have the problem?

EA5SW, 2021-03-11:
> @EA5SW
> 
> Is V2 perfect ? Or does it still sometimes have the problem?

V2 don't have any problem for me, V3 have sometimes periods of mute audio.

I flash again V2 to retest

EA5SW, 2021-03-13:
I will try to explain how my GD-77 behaves after the various updates.

As we already know from the versions with level 4 echo, some GD-77s stopped receiving DMR audio or sounding the beeps after a few seconds, or minutes, the bug was random and without a defined time.

The version that has always worked well has been the one that does a full reset to the C6000 processor.

After them we have tried a few versions from send 0xaa, of which V2 was the one that gave a similar result to full reset.

When I installed V3 we had errors again, the beep stopped working and the audio of many DMR conversations disappeared.

I have reinstalled the V2 to observe that the errors of the V3 have not disappeared when updating, even having done a reset with Sk2 and power.

I have re-flashed the full reset version and everything has returned to normal.

It should go back to 0xaa and then to V2 to confirm that everything is ok.

All the tests I am doing now with eco level 4 to try the maximum battery saving.

If you want me to do some additional testing, as I am always at your disposal, I already know that it is a very small number of GD-77 that create these errors, but in case it may be useful for any future changes.

rogerclarkmelbourne, 2021-03-13:
@EA5SW 

Unfortunately I think it is impossible to have the best power savings on radios which have the old C6000 chips with hardware bugs in them.

I added the code to send 0xAA on all Eco levels and this seems to completely fix the problems for some people but not your radio.

Eco Level 1 does not power down the C6000, so should not have any bugs in your radio, apart from there may still be problems if you attempt to use beep volume levels less than -12dB, because it appears that if the output DAC in the C6000 has long sequences of low value audio, the DAC stops working completely, until many 0xAA are sent to the DAC.

The official firmware does not have these problems because it never has low beep levels, and the official firmware does not use the sleep mode of the C6000.

However because the C6000 consumes 25% of the current of the radio, so its important that I use the sleep function so that 99.9%  which don't have faulty C6000 chips.  get much better battery life.