rogerclarkmelbourne/OpenGD77 #916: Tone scan doesn't work when Filter is set to None #2

dmitrybabeshko, 2021-03-06:
Hello Roger,

This is a follow-up for #910 since the original has been closed.
The issue is still reproducible with the latest development release D2021.03.06.01 on DM-1801.

Thank you in advance.

f1rmb, 2021-03-06:
Hi,

  It seems the fix hasn't been merged yet.

Thanks for the notice.
...
Daniel

rogerclarkmelbourne, 2021-03-06:
Sorry.

Somehow that test branch didn't get merged back into my local main version.

I've merged it now into my local copy

[OpenGD77_20210307_fix_916.zip](https://github.com/rogerclarkmelbourne/OpenGD77/files/6096114/OpenGD77_20210307_fix_916.zip)

dmitrybabeshko, 2021-03-07:
> I've merged it now into my local copy
> [OpenGD77_20210307_fix_916.zip](https://github.com/rogerclarkmelbourne/OpenGD77/files/6096114/OpenGD77_20210307_fix_916.zip)

Checked this build. Issue is fixed. Thank you.

dmitrybabeshko, 2021-03-24:
Checked https://github.com/rogerclarkmelbourne/OpenGD77/releases/tag/D2021.03.21.01
Issue is resolved

rogerclarkmelbourne, 2021-03-24:
OK

closed 2021-03-24 by rogerclarkmelbourne