rogerclarkmelbourne/OpenGD77 #919: Development branch

on7lds, 2021-03-07:
I want to update the Dutch translations, but the development branch is missing ?
Or am I overlooking something ?

f1rmb, 2021-03-07:
You need to use the master branch, the development one has been removed.

on7lds, 2021-03-07:
* With that branch I do not seem to have the new menu's (radio info, temperature ...) ?
* I do the PR then to the master branch ?

f1rmb, 2021-03-07:
The master has these entries. The only missing ATM are .buttons and .leds.
Yes, you should raise PR against master branch.

on7lds, 2021-03-07:
Compiling the Github tree gives me the menu 'Battery' without the temperature.
Flashing the  (only possible choice) 'unstable' firmware with the OpenGD77CPS gives me the 'Radio info' menu with temperature.
What am I doing wrong ?

rogerclarkmelbourne, 2021-03-07:
@on7lds 

The public repo does not contain the latest features.

It receives bug fixes and language updates but not major feature updates at the moment.

on7lds, 2021-03-08:
Ah, OK.
Then I'll already just translate and wait for the functions to come in the master branch to verify if they fit (they should, but I like to verify).

note: README.md, still says to PR against the development branch.

rogerclarkmelbourne, 2021-03-09:
Readme has now been updated

closed 2021-03-09 by rogerclarkmelbourne