rogerclarkmelbourne/OpenGD77 #925: Flashlight (LED) sk2+ 5 no longer working

VA3NDP, 2021-03-09:
RD-5R flashlight function no longer works when I push function button 2 + #5. Stopped working after last update. 
Current firmware- be33e68 
March 6, 2021

rogerclarkmelbourne, 2021-03-09:
OK.
There were no specific changes to the buttons in the last update.,

The bug was probably introduced some time ago but remained unnoticed

rogerclarkmelbourne, 2021-03-09:
Ah,

I wonder if this is something to do with the changes to the LEDs

VA3NDP, 2021-03-09:
Quite possible. I rarely use it but I did notice it and thought I should note it. It's definitely not a big deal but I thought you would like to know. 
I did operate the new LED on/off function in the menu to see if that made a difference but no change.

rogerclarkmelbourne, 2021-03-09:
I just tried my RD5R and the light is working fine.

Which version are you using ?

VA3NDP, 2021-03-09:
Yup. Beta March 6

rogerclarkmelbourne, 2021-03-09:
I just loaded that, and my RD5R LED still works as expected.

Which version still works OK for you ?

VA3NDP, 2021-03-09:
The one directly ahead of that one was just fine. Instead of downgrading I can wait until the next release and see how it is? It's not the end of the world :)

VA3NDP, 2021-03-09:
Did you want me to try a re-load or different version?

VA3NDP, 2021-03-09:
I went back 5 firmwares and tried all of them with no luck. I'm thinking maybe my light is toast and I wasted your time? It's only 2 months old but it's possible ?? I can't wait until my 1801/860 arrives in a couple weeks 😉

rogerclarkmelbourne, 2021-03-09:
Yep. Sounds like the LED packed up.

The RD5R hardware is junk really. Its only supported because some guy in Russia did the original conversion for the DM-5R V2.

But its been nothing but trouble ever since.

rogerclarkmelbourne, 2021-03-09:
OK

closed 2021-03-09 by VA3NDP