rogerclarkmelbourne/OpenGD77 #926: Display corruption (DM1801) Zone screen - reported on the forum

rogerclarkmelbourne, 2021-03-11:
Reported

https://www.opengd77.com/viewtopic.php?f=10&t=1564

Unconfirmed.

From version 2021.02.13.01 reaching the "Zone" menu using the "Quick Key" function, display appears OK but scrolling beyond the first top line or beyond the last bottom line I got corrupted lines with strange symbols or blank lines. Sometime, if I select a corrupted line then the "Zone" menu will be corrupted even when I reach it through the MENU button.
To fix the issue I must scroll through zone or switch the radio Off/On.
No issue using the MENU button to reach the Zone menu.

f1rmb, 2021-03-11:
I know why, fix is on its way.

rogerclarkmelbourne, 2021-03-11:
Fix from Daniel

[OpenGD77_fix_bug_926.zip](https://github.com/rogerclarkmelbourne/OpenGD77/files/6121807/OpenGD77_fix_bug_926.zip)

closed 2021-04-15 by rogerclarkmelbourne