rogerclarkmelbourne/OpenGD77 #927: Possibly remember quick menu last position (not saved to nonvolatile storage)

rogerclarkmelbourne, 2021-03-11:
https://opengd77.com/viewtopic.php?f=7&t=1566&p=13853#p13850

This probably also applies to the main menu.  I think the official firmware remembers the main menu position while the power to the radio is on.

g4eml, 2021-03-11:
It probably makes sense to apply the same logic to all menus including things like the channel details. If I have just changed something it is quite likely that I will want to return to it and change it again.

rogerclarkmelbourne, 2021-03-11:
We could store this in RAM, but storing all menus in the EEPROM won't fit in the single page (128 bytes) as I'm pretty sure we are already using 120 + bytes.

We don't want to store more than 1 page of data to the EEPROM as it would require 2 writes and on the RD5R it has to write the data ever time the user makes a change to the radio, and although we defer saving the change for a short while, so that the EEPROM does not get worn out by loads of unnecessary writes, having to write to 2 separate pages will double the processing.

g4eml, 2021-03-12:
I don’t think they need to be saved to EEPROM.  Just storing in RAM would be a worthwhile improvement.

f1rmb, 2021-03-12:
Hi,

I have something about this.
Of course, it won't survive a power cycle.

rogerclarkmelbourne, 2021-03-12:
OK.

I did not have time to investigate this.

It would need an array of values, i.e one per menu index, plus separate support for the quick menu's
(Depending on which menus remember their position)

f1rmb, 2021-03-12:
My implementation is generic, all menu are handled the same way, by the menu system layer.
Wait a sec, I have a PR ;-)

rogerclarkmelbourne, 2021-03-12:
No rush....

rogerclarkmelbourne, 2021-03-13:
Feature will be in the next Beta release