rogerclarkmelbourne/OpenGD77 #930: Translating to Russian TAKE#2

niaproxy, 2021-03-15:
Hi!
I made some effort in russian translation: https://github.com/niaproxy/OpenGD77
What's done:
1. Translated 95% of text. (Need some correction on used terms and abbreviations but nothing big)
2. Made custom cyrillic fonts for 8x8 layout

Code compiles, radio(RD-5R) boots, all text strings on places.
Currently to enable translation uncomment define in uiLocalisation.h

f1rmb, 2021-03-15:
Hi,

  I looked at your repo.
There are few error in the charset file (duplicated and unsupported encoded sequences).

So, I'm attaching a fixed version:
FONT_CHAR_RUS_CAPITAL_CE was badly encoded
FONT_CHAR_RUS_CAPITAL_SH was duplicated, second one was renamed FONT_CHAR_RUS_CAPITAL_SH2
FONT_CHAR_RUS_SMALL_SH was duplicated, second one was renamed FONT_CHAR_RUS_SMALL_SH2

Also, do not use glyph @ 176 offset, it's Celcius degree, and we use it.

It seems you're editing the fonts by hand. I can send you a tarball with all the splitted images (png), for all the fonts.
And finally, if you want to have a change to see Russian language merged into the code, you need to draw the whole fonts set.

Cheers.
---
Daniel
[UC1701_charset_RU.h.zip](https://github.com/rogerclarkmelbourne/OpenGD77/files/6144368/UC1701_charset_RU.h.zip)

rogerclarkmelbourne, 2021-03-15:
BTW.

I presume adding Russian will need another 4 build targets :-(

If so, I think there would need to be separate project files for each font, because I need to be able to easily build for the 4 hardware platforms, to check for compile problems, so I need to use Clean All and Build All.
But to build 12 targets will take a long time and will be a complete pain.

niaproxy, 2021-03-17:
Hi! 
@f1rmb Thanks for feedback. With charset it surely my mistake...
About Celcius degree is another story. ISO-8859-5 codeset (which I used for compatibility) does not have such symbol. Instead there are cyrillic "А" and i temporary replace it by ' symbol in russian.h. Sure not the best solution.

As for rest of the work(i.e. fonts and docs), one of reasons I open issue is to gain community attention. If someone native speaking russian and will to help, it will be good starting point. 

@rogerclarkmelbourne I ask to leave issue open for a while, maybe some more people will join. 
I also publish binary release for RD-5R on my fork page if someone curios.

For russian friends:
Ссылка на мой форк с собраной прошивкой для RD-5R в шапке. Желающим помочь с переводом предлагаю присоединяться.

Best luck!

niaproxy, 2021-03-17:
Also i notice something weird in language menu. Not sure why it appears...
[image]

f1rmb, 2021-03-17:
Hi,

> Hi!
> @f1rmb Thanks for feedback. With charset it surely my mistake...
> About Celcius degree is another story. ISO-8859-5 codeset (which I used for compatibility) does not have such symbol. Instead there are cyrillic "А" and i temporary replace it by ' symbol in russian.h. Sure not the best solution.

I think you missed the point of the OpenGD77's binary encoding ;-)
The remapping you defines in the charset file is used by the UC1701_font_tool to convert language filles UTF <-> binary formats.

So, the original RU translation file has to be in UTF-8 encoding, and the output will be binary, with remapped to correct font offset. From that binary encoded file, it can rebuild the UTF-8 version.

But you still need to draw the other 4 missing fonts.

Cheers.
---
Daniel

closed 2021-04-20 by rogerclarkmelbourne