rogerclarkmelbourne/OpenGD77 #932: GD77S filter mode not accessible

rogerclarkmelbourne, 2021-03-23:
Bug report on the forum 

https://www.opengd77.com/viewtopic.php?f=13&t=1593

The Filter mode is being bypassed in both DMR and FM modes.

I will post a fix soon.

rogerclarkmelbourne, 2021-03-23:
[OpenGD77_FixGD77S_bug_932.zip](https://github.com/rogerclarkmelbourne/OpenGD77/files/6187168/OpenGD77_FixGD77S_bug_932.zip)

f1rmb, 2021-03-23:
Hi Roger,

  I was also fixing this ;-)

rogerclarkmelbourne, 2021-03-23:
OK.

I thought if I make a bug and assign it to me, then you would not waste your time on it.

f1rmb, 2021-03-23:
Ah, no worries. I just saw the post few minutes ago ;-)

I will raise a PR to this branch, for related bugfixes/changes.

rogerclarkmelbourne, 2021-03-23:
OK. I pushed a fix, but perhaps there are other problems

f1rmb, 2021-03-23:
I think this bug is there since the filtering has been splitted (yeah, that's pretty old ;-) ).

rogerclarkmelbourne, 2021-03-23:
[OpenGD77_FixGD77S_bug_932_v2.zip](https://github.com/rogerclarkmelbourne/OpenGD77/files/6187495/OpenGD77_FixGD77S_bug_932_v2.zip)