rogerclarkmelbourne/OpenGD77 #936: Battery calibration

ZZyZer, 2021-04-04:
Hi!
Is it possible to add "Battery calibration" in Options, just like "Temperature calibration"? My DM-1801 is showing wrong battery voltage. So fully charged battery 8.35v is displayed as 8.0v (83% charge).
Thanks a lot!

rogerclarkmelbourne, 2021-04-04:
You mean to compensate for radios with faults in the hardware or the battery ?

ZZyZer, 2021-04-04:
Yes, that's right. 
I agree it is bad idea to compensate faulty battery . But compensate wrong tuned hardware should be acceptable.

ZZyZer, 2021-04-04:
I'll try to voltage a bit later. I'm not good at electronics, but if you provide a picture - it will help a lot.

rogerclarkmelbourne, 2021-04-04:
Its very difficult to read the internal voltage.

I think there may be a hardware design error in the GD77, and possibly the same mistake in the DM-1801, where there is a 47k resistor in series with the ADC input. But normally the input impedance of ADC's is not super high, perhaps 100k or 200k, so 47k in series will affect the values :-(

However my DM-1801 works OK. Battery is 8.4V and the radio displays 8.4V

My GD-77 are less accurate and read about 0.2V low, allowing for granularity

rogerclarkmelbourne, 2021-04-04:
Daniel created a version with some experimental battery calibrations

See Options -> Battery
Range is +/- 0.5V in 0.1V steps

[OpenGD77_202104042114_Bat_CAL_TEST_V1.zip](https://github.com/rogerclarkmelbourne/OpenGD77/files/6254424/OpenGD77_202104042114_Bat_CAL_TEST_V1.zip)

ZZyZer, 2021-04-04:
Thank you very much!
At fully charged battery 8.32v battery radio displays 8.3v at 100%
I'll continue testing at various battery voltage and report later.
Thank you once again!

rogerclarkmelbourne, 2021-04-04:
I have tested a GD77 and a DM1801 from 8.4V to 6.2V and the readings were accurate to 0.1V

Initially I thought there may be a need for a scale factor as well as an offset, but for the limited voltage range over which the radio can operate, a scale factor does not appear to be necessary.

There will always be the possibility of a difference of 0.1V between the measured voltage and the displayed voltage, because of the granularity of the display, but the firmware is not intended as a volt meter for the battery, and Daniel and I have noticed hardware design errors in the battery measurement circuit of the GD77.

I don't know if those errors exist on the DM1801 as no one has made a schematic for that radio, but its probably likely that Baofeng copied the bad design from the GD77

ZZyZer, 2021-04-04:
Most likely, Baofeng copied the early design of GD77 radio. Perhaps in later versions this defect was corrected.
About the calibration. Calibration works well over the entire voltage range of the battery. I really hope that this option will appear in future firmware releases.
Thanks a lot again!

rogerclarkmelbourne, 2021-04-04:
It will be in the next Beta

Reednier42, 2021-04-05:
I can report similar issue with DM-1801. Previously my OpenGD77 worked fine (-0.1V error) what comes to the battery reading, but my radio got bricked and I had to recover it from the backups.  Now i's running  April 3th 2021 firmware and battery reading says 9V with full 8.3V battery and 7V at the low battery  so it's off +0.7V. (measured from the battery pins) .  I tried to roll back to previous firmware, but problem was there so I came back to current development release.   I hope the Bat Cal adjustment value could be a bit wider to cover worst case offsets, something like +-1V  with 0.1V steps.

f1rmb, 2021-04-05:
@Reednier42 did you restore your calibrations too ?

f1rmb, 2021-04-05:
BTW, can you post your calibration here ?

rogerclarkmelbourne, 2021-04-05:
Calibration should not effect the battery, its only for RF.

@Reednier42 

How did the radio get bricked?  Are you sure you didn't somehow damage the hardware?

Reednier42, 2021-04-05:
My radio just went dark when I was trying to load in a user ID database.  Not sure what happened, maybe I tried to put too large data base into the radio memory and something when into the wrong place.  Result was the Low BT version text and silent radio.  I looked recovery instructions from the forum and restored my radio from backups Flash, Cals, code plug, etc. After that I updated the radio to latest development version and noticed that I had  battery voltage, temperature and frequency off set completely off the chart, all the other parameters were fine.  I have now calibrated my radio back to frequency against a radio tester and now it seems to work fine, only battery and temperature are off the chart.  Not sure if there are any model differences, but my DM-1801 came with older factory firmware from 2018.

rogerclarkmelbourne, 2021-04-05:
@Reednier42 

What firmware were you running in the radio when it went dark?
The "Low BT" message is part of the official firmware. I didn't think the radio can display this message when running the OpenGD77 firmware, unless perhaps the bootloader in the DM1801 displays that message

Reednier42, 2021-04-06:
I think it was running19th of Jan 2021 development release when it suddenly radio died and when it did it, it flashed that Low BT message in the screen.   I try to document things better next time to make developers work more easier.

rogerclarkmelbourne, 2021-04-06:
I checked the CPS and it has a check for the maximum length of the DMR ID data, so that should not have caused the problem.

I'll need to check the Low BT message, as I thought that was in the main official firmware, if you were using the version from 19th Jan, it must be in the official bootloader

rogerclarkmelbourne, 2021-04-06:
@Reednier42 

When you say Low BT, is this the actual text ?

I checked the DM18001 firmware file and it does contain that text, but only in a block of random data included from the official firmware, and it should never access that text.

Reednier42, 2021-04-06:
Not sure if this helps, but I checked my first MCU backup file and it contained a text "no app software low bt version  no activation   successful"

rogerclarkmelbourne, 2021-04-15:
@Reednier42 

I checked an that text is because 30% of the binary is from the offcial firmware, and although that text is not used by the firmware, it is in the binary block.

There is a large portion of the binary block which could be zero'ed to prevent confustion, but it would take a long time to do this, for no appreciable overall gain..

The best thing would be to dissassemble the binary blocks and change them to remove unused junk and reassemble them, but that would also be non-trivial

Reednier42, 2021-04-18:
A voice of warning.  After testing OpenGD77_202104042114_Bat_CAL_TEST_V1.zip  in my DM-1801 I noticed that I could adjust battery level, but side effect was that radio side key operation went weird.  PTT stopped working  Orange key, S1 and Fn trigger PTT and it's stuck. I tried to restore flash & eeprom backups and then re-flash for latest version, but it did not help.  I went far as going back to original software 2.2.5, but it was stuck it the TX mode automatically after the bootup.  Now I'm back in OpenGD77 and radio seems to be is still usable in hotspot mode.

ZZyZer, 2021-04-19:
I have no negative effects. The radio works fine without any problems. Battery calibration also works well.

Reednier42, 2021-04-19:
The problem maybe in my radio, maybe it has a faulty memory block or I just managed to push right combinations to trigger a bug that affects memory outside the OpenGD77. I just know that this side button problem started after I had updated the firmware and adjusted the radio settings to my liking.  

If recall right I changed battery level -0.5, temperature to -10, VHF and UHF squelch 30%, Hotspot MMDVM,  TA Tx on, Ecolevel to 0, Display brightness 20%, etc. So a lot of changes to the settings before I noticed that TX gets stuck on from other side buttons than PTT,  PTT button itself seems to be completely silent.

closed 2021-04-20 by rogerclarkmelbourne