rogerclarkmelbourne/OpenGD77 #937: Frequency range in VFO A seems to be limited but VFO B seems ok

vk7js, 2021-04-05:
Hi Roger/Daniel,
I have just noticed in the last firmware prior to battery calib being added, that the vfo rx freq cannot be set above 150 MHZ in VFO A. I tried to set the rx freq to 152.5 but get the error tones. It works in VFO B but not A for some reason. Is VFO A limited for a reason? If so, what are the band limits for VFO A compared to B? I checked the online manual but no limits appear to be imposed making me think this is a bug. I'm not sure if it is new, I haven't gone back further to check.
Joe
vk7js

vk7js, 2021-04-05:
Actually, it is worse than first thought. I can copy a channel to VFO A using the quick menu, but entering any frequency, even in the HAM band into VFO A yields the error tone. In fact direct frequency entry seems broken completely... Now that I try this in VFO B, it also seems hit and miss. If I either enter something like 147 hear point, then type 000, or if I enter 147 and hit green, it doesn't seem to work.

vk7js, 2021-04-05:
Now that I've rebooted my radio, things seem to be working again. I don't know how it got into this strange mode, all I can say was I did a scan in VFO mode and then tried entering a few frequencies, changing from VFO A to B, entering more frequencies, etc. I then finally got frustrated and rebooted the radio and all seems well. Please look out for weird behaviour with direct frequency entry after scanning in VFO mode.

vk7js, 2021-04-05:
Ok, after more testing it does seem that scanning in VFO messes up direct frequency entry in that VFo, which is why it seemed to work in VFO b and not a because I had performed a scan in a but not b.

f1rmb, 2021-04-05:
I'm unable to reproduce this here.
But, once you have engaged the VFO scan mode, you have to leave it (long press Down) before being able to enter a direct frequency, otherwise you're just entering the low and high frequencies boundaries.

vk7js, 2021-04-05:
Hi Daniel,

Pressing and holding down arrow does not seem to always reliably cancel the mode for me. It also gives no confirmation beep but starts repeating the frequency so you don't know what is going on. It would be good if it was smart enough that if you only entered a single frequency and hit the green button, that it would automatically cancel the range search. Also, going to channel mode and back to VFO mode should cancel any range search. THis is very puzzling otherwise.
Joe

vk7js, 2021-04-05:
I think part of the issue is that one does not know if one is in scan or direct freq entry mode. If you could do one or both of the following it might help this confusion:
1. Cancel range entry if a single valid freq is pressed and confirmed with the green button, and also  cancel when going to channel and back to VFO mode, or
 2. when in scan entry mode, perhaps instead of saying VFO A or VFO B, say VFO A Scan Range Entry, or VFO A Range Entry, or something similar. Just saying scan might be also confusing because one might think a scan is in progress.
 3.
Thanks
Joe

closed 2021-04-20 by rogerclarkmelbourne