rogerclarkmelbourne/OpenGD77 #940: Iusse with external mic on DM-1801 Baofeng

cloto3000, 2021-04-09:
Hi,
i try to use the external mic on my DM-1801. When I use the DMR mode it happens that when I press the PTT button the radio opens and closes the TX continuously not allowing me to speak.
In analog mode, however, it works perfectly. So it's a software problem and not a hardware problem. The microphone is original BAOFENG.
Best regards
Raffaele

f1rmb, 2021-04-09:
Hi,

   There is too few details (e.g. no firmware version).
Also, did you checked the forum about this topic ? (like https://www.opengd77.com/viewtopic.php?f=7&t=1271&p=11182&hilit=external+mic&sid=1ec34efce5cf61ba18c2f629a6013a28#p11182).
FYI, In analog, there is no RF pulses.

Cheers.
---
Daniel

cloto3000, 2021-04-09:
Hi,

my problem i different. In Analog mode works fine (FM). Meanwhile in DMR mode when I push on PTT the radio ON and OFF TX mode every 0,5 Sec. 

The problem was with the prerelease software on 21 th march.

Now I have install the ultimate software RELEASED (I attach it) the radio now works also in DMR mode without problem !!

I hope to help you with the new version.

Cheers

 

Raffaele

A: rogerclarkmelbourne/OpenGD77
Cc: cloto3000; Author

 

Also, did you checked the forum about this topic ? (like https://www.opengd77.com/viewtopic.php?f=7 <https://www.opengd77.com/viewtopic.php?f=7&t=1271&p=11182&hilit=external+mic&sid=1ec34efce5cf61ba18c2f629a6013a28#p11182> &t=1271&p=11182&hilit=external+mic&sid=1ec34efce5cf61ba18c2f629a6013a28#p11182).

rogerclarkmelbourne, 2021-04-09:
Check the wiring of the microphone.

Some microphones have the PTT connection routed through the speaker, casuing 8 ohms resistance in series with the PTT circuit.

On DMR RF pickup into the cable and the speaker, injects pulses into the PTT input which cause the transmission to terminate then restart aka "machine gunning"

This does not happen on FM because the RF is not pulsed

cloto3000, 2021-04-10:
#response_container_BBPPID{font-family: initial; font-size:initial; color: initial;}  Ok thanks you.I solved i change the mic with a Retevis. It works fine !Many thanks !Cheers                                                                                                                                                                                                                                                                                                                        Da: ***@***.***Inviato: 10 Aprile 2021 00:29A: ***@***.***Rispondi a: ***@***.***Cc: ***@***.***; ***@***.***Oggetto: Re: [rogerclarkmelbourne/OpenGD77] Issue with external mic on DM-1801 Baofeng (#940)  
Check the wiring of the microphone.
Some microphones have the PTT connection routed through the speaker, casuing 8 ohms resistance in series with the PTT circuit.
On DMR RF pickup into the cable and the speaker, injects pulses into the PTT input which cause the transmission to terminate then restart aka "machine gunning"
This does not happen on FM because the RF is not pulsed

closed 2021-04-11 by rogerclarkmelbourne

cloto3000, 2021-04-11:
Hi Daniel,

the issue was resolved. The problem was the Baofeng MIC that create RF when push PTT. Now I use a Retevis and works fine.

Best regards

Raffaele

rogerclarkmelbourne, 2021-04-11:
You can probably fix the Beofeng mic if you dissassemble it and modify the wiring