rogerclarkmelbourne/OpenGD77 #941: rd5r hotsport

eedgf, 2021-04-13:
HAM be free to choose。thanks！

f1rmb, 2021-04-13:
It's not a matter of being an Ham or not.
RD5R just kills the USB connection when it starts transmitting, so hotspot doesn't work, and couldn't work.

rogerclarkmelbourne, 2021-04-14:
The hotspot functionality was removed from the RD5R on 2020/07/24 (24th July 2020)

You install the version prior to its removal.

https://github.com/rogerclarkmelbourne/OpenGD77/releases/tag/R2020.05.03

However, it's not going to work reliably, as the hardware in the radio has very bad internal RFI and supply regulation problems, which cause the USB bus to fail frequenctly when the radio transmits.

The hotspot may work for a while but it will not be reliable, even on very low power settings 50mW

The only way to make the hotspot work with the RD-5R is to modify the hardware to improve the RFI and supply voltage problems.  I know a few people did that, and can then use this radio as a hotspot, but it is not reliable unless you modify the hardware.

If you just want a radio as a high power hotspot, buy a GD-77S, or DM-1801 or GD-77.

eedgf, 2021-04-14:
Thank you very much for your reply, because I have two DM5Rs, and one dm5r can work normally with hotstop, so I hope to be able to choose the hotstop function freely.
I also learned that some dm5r hotstops can work normally.

rogerclarkmelbourne, 2021-04-14:
I think its unlikely your radios will work as a hotspot, I think when people made the RD-5R work as a hotspot they made some changes, perhaps special USB cables, special power supply (not battery), only use external antenna etc etc

eedgf, 2021-04-14:
Thank you very much for your patient reply. I am very sorry for taking up your precious time. Can you provide dm5r to support hopstop to compile the code or configuration that needs to be modified?

rogerclarkmelbourne, 2021-04-14:
This isn's a commercial product, so there is no "support"

You can compile this codebase yourself, and modify the code to allow the hotspot 

The functionality was disabled by this commit 

https://github.com/rogerclarkmelbourne/OpenGD77/commit/1822942037f3fed8ae471c1019c76b0801421134#diff-31e12a76636d88dd49594b7b88c24d5f522e7193f8fefe4b7bb1e80ae739d7dd

Details how how to rebuild the firmware are in the wiki

https://github.com/rogerclarkmelbourne/OpenGD77/wiki/OpenGD77-SW-Development

The instructions about how to build are for developers, there is not a simple guide because development of embedded firmware is non-trivial and generally if you don't understand how to install the development enviroment and build the firmware, you are very unlikely to be able to make useful changes

eedgf, 2021-04-14:
thank you very much

closed 2021-04-14 by eedgf